EFForg/crocodilehunter #83: Experiment with using Mozilla Location Service to detect new cell towers

simonft, 2020-06-20:
[Mozilla Location Service](https://location.services.mozilla.com/) has some cell sites that wigle and opencellid don't, at least in NYC. It doesn't look like there's a good way to search for a specific cell tower using the API, but they do provide gzipped versions of the full data here: https://location.services.mozilla.com/downloads. 

We could either build and run our own API to query the data or download it and ingest into either the mysql database we already have or a sqlite db.

The sqlite db seems like the easiest version. Building and running an API is probably more work than it's worth before we know it's useful, and importing into the mysqldb means we'd be storing a whole copy for each individual project.

marcfielding1, 2020-08-10:
I can spin you guys up a crud API on AWS Lambda if it helps, once I understand the requirements it'd like be a days work.

cooperq, 2020-08-14:
that would be rad! definitely want to check as many reference DBs as possible. @marcfielding1 any idea how much work and $$ that would be?

marcfielding1, 2020-08-16:
@cooperq Depends what you need if it's just a couple of endpoints you can just buy me a beer if you ever come this way. You can get me on LinkedIn here: https://www.linkedin.com/in/marc-fielding-a8bb3293/ - if you like we can arrange a zoom call and I'll get it spun up within a day I reckon.

cooperq, 2020-08-25:
DM me on twitter and we can chat, same username as here: @cooperq

simonft, 2020-09-13:
I'm not sure if the two of you have talked on twitter, but my thoughts:

I'm not sure if we need a fancy searching api. I think we're just going to be searching by tower id and cell id, and Cooper correct me if I'm wrong but I don't think we need/want to return anything if only one of them matches. It's probably good enough to use a key/value store where the key is "{tower_id}_{cell_id}". Then if the tower isn't in the database, or if it's only been seen far away in the past, it can be flagged. For that we can probably get away with just using say an s3 bucket. Either it can require an AWS access key to use, or we can just leave it open. A malicious actor would have to make quite a few requests to it before the amount of money starts to get noticeable.