EFForg/crocodilehunter #86: test with limesdr and other sdrs supported by soapysdr

cooperq, 2020-07-20:
Test to see if crocodile hunter can run on a cheaper SDR:
* Lime SDR
* HackRF
* RTL-SDR

phantomdennis, 2020-08-06:
How do I sign up to test with a HackRF?

alphafox02, 2020-08-08:
Not necessarily cheaper but the usrp b205mini-I works with uhd. Confirmed, at least in my case, that it works. Is there a way to pass a soapy argument to srsue via the crocodile py file?

aphid, 2020-08-09:
I have a lime SDR but would need some guidance to test

pdelteil, 2020-08-12:
I have the three devices. I will give it a try. Where should I report the results ?

jofrep, 2020-08-14:
> 
> 
> I have a lime SDR but would need some guidance to test

me too. I have lime SDR and would like to test but would need some guidance

cooperq, 2020-08-14:
You will have to start by enabling the SOAPY SDR drivers in https://github.com/EFForg/srsLTE/blob/master/CMakeLists.txt and then get your device working with SOAPY SDR. Compile SRSLTE by hand then test it with one of the test scripts. If it all works please let me know here!

alphafox02, 2020-08-15:
https://youtu.be/AbtmqkFkmIg 

Made this for someone else trying to use soapy and srslte. I don’t really have an issue using soapy so long as there’s only one device plugged in and/or being recognized by soapy as I don’t know yet how to pick individual devices when using siapy with srslte. It just picks the first device in the list so make sure you have only one thing being recognized by SoapySDRUtil, otherwise if you know the arguments for add for specific devices please share. 

Also, which soapy works when using some things with srslte directly, not sure how to pass this along when starting it with the crocodile file.

phantomdennis, 2020-08-17:
So I have SoapySDR running with the HackRF. It looks like it is found without issue. 
```
ubuntu@ubunu2004:~/srsLTE$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Found device 0
  device = HackRF One
  driver = hackrf
  label = HackRF One #0 XXXXXXXXXXXXXXX
  part_id = a000cb3c0055434e
  serial = 0000000000000000XXXXXXXXXXXXXXX
  version = 2018.01.1
```

I made sure that the SoapySDR was enabled in the CMakeLists.txt file for srsLTE.

```
option(ENABLE_GUI      "Enable GUI (using srsGUI)"                OFF)
option(ENABLE_UHD      "Enable UHD"                               ON)
option(ENABLE_BLADERF  "Enable BladeRF"                           ON)
option(ENABLE_SOAPYSDR "Enable SoapySDR"                          ON)
option(ENABLE_HARDSIM  "Enable support for SIM cards"             OFF)
``` 
When running the crcodilehunter.py I get the following output. Is there something that needs to be passed to tell the script to use SoapySDR?

```
ubuntu@ubunu2004:~/Desktop/crocodilehunter/src$ ./crocodilehunter.py -p default
* 22:00:10 default - INFO starting crododile hunter project: default
 * 22:00:10 default - INFO Starting WebUI
 * 22:00:10 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 22:00:10 default - INFO Checking for config file
 * 22:00:10 default - INFO Checking for SDR
 * 22:00:10 default - INFO No bladeRF devices connected. Assuming UHD device.
 * 22:00:10 default - INFO No UHD device found either
 * 22:00:10 default - CRITICAL Bootstrapping failed
 * 22:00:10 default - ERROR Exiting...
 * 22:00:10 default - WARNING Stopping Watchdog
 * 22:00:10 default - SUCCESS See you space cowboy...
```

cooperq, 2020-08-17:
ah yea the problem here is the bootstrap script I have looks for either a  bladeRF or Ettus device and quits if it doesn't find one. If you know bash this should be pretty easy to patch, if not I can try to get a patch today . 

If you run `crocodilehunter/src/srsLTE/build/lib/examples/cell_measurement -z <list,of,earfcns>` on its own with the hackrf does it work?

cooperq, 2020-08-17:
closing #91 as a duplicate of this issue.

phantomdennis, 2020-08-17:
Yeah, I noticed that. I commented the tests out and found an issue with my gps. I'm working on that at the moment. I'll let you know if I get to the point where I can build and test the hackrf with the cell_measurement. Hopefully soon.

phantomdennis, 2020-08-17:
So I was able to get it to run by commenting out the bladeRF and Ettus device checks. The web interface starts but still no joy on the SoapySDR for HackRF.

```
* 00:09:20 default - INFO starting crododile hunter project: default
 * 00:09:20 default - INFO Starting WebUI
 * 00:09:20 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 00:09:20 default - INFO -e Checking for config file
 * 00:09:20 default - INFO -e Checking for SDR
 * 00:09:20 default - INFO -e Starting GPS
 * 00:09:21 default - INFO Warning: Stopping gpsd.service, but it can still be activated by:
 * 00:09:21 default - INFO   gpsd.socket
 * 00:09:21 default - INFO -e No GPS device found
 * 00:09:21 default - INFO -e Compiling srsUE
 * 00:09:24 default - INFO /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c: In function ‘rf_soapy_open_multi’:
 * 00:09:24 default - INFO /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:347:52: warning: passing argument 2 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
 * 00:09:24 default - INFO   347 |     if(SoapySDRDevice_setupStream(handler->device, &(handler->rxStream), SOAPY_SDR_RX, SOAPY_SDR_CF32, rx_channels, numChannels, NULL) != 0) {
 * 00:09:24 default - INFO       |                                                    ^~~~~~~~~~~~~~~~~~~~
 * 00:09:24 default - INFO       |                                                    |
 * 00:09:24 default - INFO       |                                                    SoapySDRStream ** {aka struct SoapySDRStream **}
 * 00:09:24 default - INFO In file included from /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:37:
 * 00:09:24 default - INFO /usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘int’ but argument is of type ‘SoapySDRStream **’ {aka ‘struct SoapySDRStream **’}
 * 00:09:24 default - INFO   295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
 * 00:09:24 default - INFO       |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
 * 00:09:24 default - INFO /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:347:74: warning: passing argument 3 of ‘SoapySDRDevice_setupStream’ makes pointer from integer without a cast [-Wint-conversion]
 * 00:09:24 default - INFO   347 |     if(SoapySDRDevice_setupStream(handler->device, &(handler->rxStream), SOAPY_SDR_RX, SOAPY_SDR_CF32, rx_channels, numChannels, NULL) != 0) {
 * 00:09:24 default - INFO       |                                                                          ^~~~~~~~~~~~
 * 00:09:24 default - INFO       |                                                                          |
 * 00:09:24 default - INFO       |                                                                          int
 * 00:09:24 default - INFO In file included from /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:37:
 * 00:09:24 default - INFO /usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const char *’ but argument is of type ‘int’
 * 00:09:24 default - INFO   295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
 * 00:09:24 default - INFO       |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
 * 00:09:24 default - INFO /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:347:88: error: passing argument 4 of ‘SoapySDRDevice_setupStream’ from incompatible pointer type [-Werror=incompatible-pointer-types]
 * 00:09:24 default - INFO   347 |     if(SoapySDRDevice_setupStream(handler->device, &(handler->rxStream), SOAPY_SDR_RX, SOAPY_SDR_CF32, rx_channels, numChannels, NULL) != 0) {
 * 00:09:24 default - INFO       |                                                                                        ^~~~~~~~~~~~~~
 * 00:09:24 default - INFO       |                                                                                        |
 * 00:09:24 default - INFO       |                                                                                        char *
 * 00:09:24 default - INFO In file included from /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:37:
 * 00:09:24 default - INFO /usr/local/include/SoapySDR/Device.h:295:31: note: expected ‘const size_t *’ {aka ‘const long unsigned int *’} but argument is of type ‘char *’
 * 00:09:24 default - INFO   295 | SOAPY_SDR_API SoapySDRStream *SoapySDRDevice_setupStream(SoapySDRDevice *device,
 * 00:09:24 default - INFO       |                               ^~~~~~~~~~~~~~~~~~~~~~~~~~
 * 00:09:24 default - INFO /home/ubuntu/Desktop/crocodilehunter/src/srsLTE/lib/src/phy/rf/rf_soapy_imp.c:347:104: warning: passing argument 5 of ‘SoapySDRDevice_setupStream’ makes integer from pointer without a cast [-Wint-conversion]
 * 00:09:24 default - INFO   347 |     if(SoapySDRDevice_setupStream(handler->device, &(handler->rxStream), SOAPY_SDR_RX, SOAPY_SDR_CF32, rx_channels, numChannels, NULL) != 0) {
 * 00:09:24 default - INFO       |                                                                                                        ^~~~~~~~~~~
 * 00:09:24 default - INFO       |                                                                                                        |
[... cut ...]
 * 00:11:04 default - INFO 127.0.0.1 - - [18/Aug/2020 00:11:04] "GET /logs HTTP/1.1" 200 -
 * 00:11:06 default - INFO 127.0.0.1 - - [18/Aug/2020 00:11:06] "GET / HTTP/1.1" 200 -
 * 00:11:10 default - INFO 127.0.0.1 - - [18/Aug/2020 00:11:10] "GET /map HTTP/1.1" 200 -
 * 00:11:12 default - INFO 127.0.0.1 - - [18/Aug/2020 00:11:12] "GET /favicon.ico HTTP/1.1" 404 -
^ * 00:11:17 default - CRITICAL You pressed Ctrl+C!
 * 00:11:17 default - ERROR Exiting...
 * 00:11:17 default - WARNING Stopping Watchdog
 * 00:11:17 default - SUCCESS See you space cowboy...
```

alphafox02, 2020-08-17:
So even if you build in soapy support (I just did with no errors), if you have a look at the examples they only allow or at least say they do “ -a RF args” and not device_name which is what I think is needed. If you have a look at the UE.conf you’ll see that device_name is different than device_args. I’m trying to run any of the examples with soapy but it reverts to uhd. I’ll keep trying but I think something more is needed.

phantomdennis, 2020-08-17:
I don't think that the `crocodilehunter/src/srsLTE/build/lib/examples/cell_measurement` is being built. The directory is empty. I do have the `crocodilehunter/src/srsLTE/lib/examples/cell_measurement.cc` file.

alphafox02, 2020-08-17:
Go to the src/srslte directory. 
mkdir build 
cd build 
cmake .
make 

See where it’s failing. Are you building on 20.04? If so, it doesn’t build. 18.04 works.

phantomdennis, 2020-08-17:
I am building on 20.04. So I need to do this on 18.04? I'll get a VM put together and see if I can try that out this afternoon.

alphafox02, 2020-08-17:
I put together a video of how I built it on a project I maintain, which is based on 18.04. You could maybe use the video as a reference should you encounter the same problems as me. Later today or tomorrow I was going to just upload a dvd that has everything running by default. 

But as to 20.04. I could not get the required version of srslte to compile, however, I could get the newest srslte to compile. So I tried various changes mentioned in other issue tickets but no luck for me. 

https://youtu.be/4WyJ8oVl4bo

phantomdennis, 2020-08-18:
I used the DragonOS, thank you @alphafox02 and I was able to get the system compiled and running a bit more than before. See output below. This is with the HackRF and SoapySDR.

```
 * 11:39:28 default - INFO starting crocodile hunter project: default
 * 11:39:28 default - INFO Starting WebUI
 * 11:39:28 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 11:39:28 default - INFO Checking for config file
 * 11:39:28 default - INFO Checking for SDR
 * 11:39:28 default - INFO Starting GPS
 * 11:39:29 default - INFO Waiting for GPS to sync
 * 11:39:30 default - INFO GPS successfully got location
 * 11:39:30 default - INFO Compiling srsUE
 * 11:39:34 default - INFO SrsUE built succesfully
 * 11:39:34 default - INFO Reticulating Splines
 * 11:39:34 default - SUCCESS Watchdog server running
 * 11:39:34 default - WARNING Getting earcn list for the first time, this might take a while
 * 11:39:38 default - NOTICE Using earfcn list
 * 11:39:38 default - INFO EARFCNS: 
 * 11:39:38 default - INFO Running srsUE
 * 11:39:38 default - SUCCESS srsUE started with pid 4256
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "./crocodilehunter.py", line 154, in monitor_srslte
    line = nbsr.readline(int(self.config["general"]["crash_timeout"]))
ValueError: invalid literal for int() with base 10: '30 ;number of seconds until we assume process has crashed. On a raspberry pi this needs to be more like 120'
```

cooperq, 2020-08-18:
yea  the process is crashing but we don't know why, if you run cell_measurement on its own what is the output?

phantomdennis, 2020-08-18:
This is the output and crash report.

```
./cell_measurement -z <list,of,earfcns>

srsLTE crashed... backtrace saved in './srsLTE.backtrace.crash'...
---  exiting  ---
```
```
srsLTE.backtrace.crash

--- command='./cell_measurement -z' version=18.12.0 signal=11 date='18/08/2020 12:23:37' ---
        ./cell_measurement(+0x58128a) [0x55f8a5f4128a]
        /lib/x86_64-linux-gnu/libc.so.6(+0x3efd0) [0x7fa990dc8fd0]
        /lib/x86_64-linux-gnu/libc.so.6(+0x18e811) [0x7fa990f18811]
        ./cell_measurement(_Z10parse_argsP11prog_args_tiPPc+0x133) [0x55f8a5af57b3]
        ./cell_measurement(main+0x9a) [0x55f8a5af2c1a]
        /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7fa990dabb97]
        ./cell_measurement(_start+0x2a) [0x55f8a5af48da]
```

alphafox02, 2020-08-18:
So couple things, that exception trace back thing - it’s not actually crashing. It does it on my end and appears to be doing nothing but in fact it’s actually working and will find cells and move on. Each time it goes to look for a cell it’ll pop back up that exception thread 3 thing or more as you can sorta see in the background of the video I posted. 

As for running cell measurement on its own you need to run it something like this ./cell_measurement -z 900

Or whatever earfcn list you have.. but as I mentioned there’s no way to specific soapy and so I believe it sticks to uhd only. You can change devices args but not the name. So I think you’ll find that crocodile hunter will work for you if you had a bladerf or usrp. I’m sitting here now using usrp and seeing that same “exception in thread” rubbish you show above. No impact on it running though.

alphafox02, 2020-08-18:
I also just went ahead and added crocodile hunter to the ISO as that was quicker than making another one. I’ll upload it tonight and hopefully that’ll help some people... but I to would love this soapy + LimeSDR. or whatever. I think the examples need more added though. Hope I’m wrong.

alphafox02, 2020-08-18:
I just edited the cmake file and disabled uhd and bladerf and only enabled soapy. I’ll see what happens, maybe that’ll let us know if those examples are made to run anything but uhd.

phantomdennis, 2020-08-18:
I'm fairly certain it is using SoapySDR. I adjusted the CMakeList.txt to use only SoapySDR and turned off the UHD and USRP. I also ran the hackrf_info and the device is listed as busy while the cell_measure is running. I'm just not getting any output. What would I get if it found something with cell_measurement?

phantomdennis, 2020-08-18:
I did a cell search. An suggestion on the band? I was thinking 700mhz? I dont know if I need to put this in hz or khz

```
./cell_search -v -s 41094 -e 42000 -b 700
Opening RF device...
Soapy has found device #0: device=HackRF One, driver=hackrf, label=HackRF One #0 , part_id=a000cb3c0055434e, serial=000000000000000, version=2018.01.1,
[INFO] Opening HackRF One #0 ...
Set Rx bandwidth to 2.50 MHz
Set Tx bandwidth to 1.75 MHz
Setting up RX stream
Setting up TX stream
Available device sensors:
Available sensors for RX channel 0:
State of gain elements for Rx channel 0 (AGC not supported):
 - LNA: 16.00 dB
 - AMP: 0.00 dB
 - VGA: 16.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - LNA: 0.00 dB
 - AMP: 0.00 dB
 - VGA: 0.00 dB
Rx antenna set to TX/RX
Tx antenna set to TX/RX
Set master clock rate to 0.00 MHz
Error: Invalid band 700
Error getting EARFCN list
```

alphafox02, 2020-08-18:
Okay this is looking good at least in my test with an ettus. I deleted the build directory for srslte. I then edited the CMakeLists.txt and put OFF next to uhd and bladerf and on next to soapy. 

mkdir build 
cmake ..
make 

Now when I run the cell measurement example it tells me it’s using Soapy and not uhd!! Now the only other issue may be that I have soapy audio installed and it’s listed as the device 0 and my ettus is device 2. So with you running DragonOS do this 

SoapySDRUtil —find

If it’s listing anything other than your hackrf then remove it. 

Sudo apt-get remove soapysdr0.7-module-audio.

alphafox02, 2020-08-18:
Nice! We both did the same thing.

phantomdennis, 2020-08-18:
Bingo I can search with HackRF :)

```
./cell_search -v -s 5010 -e 5035 -b 12
Opening RF device...
Soapy has found device #0: device=HackRF One, driver=hackrf, label=HackRF One #0 , part_id=a000cb3c0055434e, serial=0000000000000000, version=2018.01.1,
[INFO] Opening HackRF One #0 
...
Set Rx bandwidth to 2.50 MHz
Set Tx bandwidth to 1.75 MHz
Setting up RX stream
Setting up TX stream
Available device sensors:
Available sensors for RX channel 0:
State of gain elements for Rx channel 0 (AGC not supported):
 - LNA: 16.00 dB
 - AMP: 0.00 dB
 - VGA: 16.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - LNA: 0.00 dB
 - AMP: 0.00 dB
 - VGA: 0.00 dB
Rx antenna set to TX/RX
Tx antenna set to TX/RX
Set master clock rate to 0.00 MHz
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
Tuned Rx to 729.00 MHz
[INFO]: Set rf_freq to 729.000 MHz
[  0/25]: EARFCN 5010 Freq. 729.00 MHz looking for PSS.

[INFO]: Setting sampling frequency 1.92 MHz for PSS search
Set Rx bandwidth to 2.50 MHz
[INFO]: Starting receiver...
[INFO]: CP-CFO: estimated=0.001197, mean=0.001197
[INFO]: PSS: id=0, peak_pos=1097, peak_value=1.006247
[INFO]: SYNC FIND: sf_idx=0, ret=0, next_state=0
```

alphafox02, 2020-08-18:
Now go back and run crocodile and just let it sit and run. This is awesome.

alphafox02, 2020-08-18:
So I get nothing now with crocodile forcing use of just soapy and the ettus. It just sits there after that value error line and if I run cell measurement on its own it starts with soapy, finds the ettus and eventually says error searching cell. If I rebuild and run with uhd it appears to run fine.

phantomdennis, 2020-08-18:
Ok, so I let it run for almost 45 minutes and I am not getting anything.

alphafox02, 2020-08-18:
I think it’s something with soapy and cell measurement because this combo with my ettus fails when running on its own or via crocodile. Works fine when not using soapy. I’ll grab a hackrf and some other hardware and continue testing. Seems encouraging that you got cell search to work.

phantomdennis, 2020-08-18:
I would love to have an ettus. Unfortunately, all I have at the moment is a hackrf and a small stack of rtlsdrs. I'm ready to test if something changes.

alphafox02, 2020-08-18:
What happens if you run cell search or measurement with the rtlsdr on a band/freq within its range?

phantomdennis, 2020-08-18:
It certainly finds the device and starts running then gets hung up.

```
./cell_search -v -s 6150 -e 6449 -b 20
Opening RF device...
Found Fitipower FC0013 tuner
Soapy has found device #0: available=Yes, driver=rtlsdr, label=Generic RTL2832U OEM :: 00000013, manufacturer=Realtek, product=RTL2838UHIDIR, rtl=0, serial=00000013, tuner=Fitipower FC0013,
Found Fitipower FC0013 tuner
Set Rx bandwidth to 0.00 MHz
Set Tx bandwidth to 0.00 MHz
Setting up RX stream
[INFO] Using format CF32.
Available device sensors:
Available sensors for RX channel 0:
State of gain elements for Rx channel 0 (AGC supported):
 - TUNER: 0.00 dB
State of gain elements for Tx channel 0 (AGC supported):
 - TUNER: 0.00 dB
Rx antenna set to RX
Tx antenna set to RX
Receiving async metadata not supported by device. Exiting thread.
Set master clock rate to 0.00 MHz
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
Tuned Rx to 791.00 MHz
[INFO]: Set rf_freq to 791.000 MHz
[  0/299]: EARFCN 6150 Freq. 791.00 MHz looking for PSS.

[INFO]: Setting sampling frequency 1.92 MHz for PSS search
Set Rx bandwidth to 0.00 MHz
[INFO]: Starting receiver...
```

cooperq, 2020-08-19:
I see this error Receiving async metadata not supported by device. Exiting thread.

I wonder if this is indicative of the hackrf just not being supported by srsLTE.

cooperq, 2020-08-19:
I'm on vacation this week but I will test the LimeSDR when I get back next week

phantomdennis, 2020-08-19:
I am able to get full scans with the latest srsLTE from the official repository. The version from this repository is not iterating through the EARFCNs like it does in the other for cell search.

alphafox02, 2020-08-20:
@phantomdennis Someone else pointed out that in the config.ini file to remove the comments so it'd look like this,

crash_timeout = 30
run_us_centeric_heuristics = false

Then run and you should find that all those errors will be gone and you'll be able to see what's going on.

phantomdennis, 2020-08-20:
@alphafox02 Good idea. I was able to run this again with those commented out and it looks like the srsUE cell_measurement is being called.

Run with a known good EARFCN tested with SDRSharp as having a reasonably good signal strength. I still get an error. I found that the SoapySDR driver fails when the srsUE fails. I have to disconnect the hackrf and reconnect because hackrf_info says the device is busy even after the process is killed. I'm getting the feeling that this not likely going to work with hackrf right this moment. I might just have to see if I can get a limeSDR mini after someone tests to see if it works.

[image]

```
* 09:13:51 default - INFO starting crocodile hunter project: default
 * 09:13:51 default - INFO Starting WebUI
 * 09:13:51 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 09:13:51 default - INFO Checking for config file
 * 09:13:51 default - INFO Checking for SDR
 * 09:13:51 default - INFO Skipping GPS test
 * 09:13:51 default - INFO Compiling srsUE
 * 09:13:55 default - INFO SrsUE built succesfully
 * 09:13:55 default - INFO Reticulating Splines
 * 09:13:55 default - SUCCESS Watchdog server running
 * 09:13:55 default - NOTICE Using earfcn list 8763
 * 09:13:55 default - INFO EARFCNS: 8763
 * 09:13:55 default - INFO Running srsUE
 * 09:13:55 default - SUCCESS srsUE started with pid 21530
 * 09:14:27 default - WARNING srsUE has exited unexpectedly
 * 09:14:27 default - WARNING It's dying words were: ***** plan dft 1d
 * 09:14:27 default - NOTICE Using earfcn list 8763
 * 09:14:27 default - INFO EARFCNS: 8763
 * 09:14:27 default - INFO Running srsUE
 * 09:14:27 default - SUCCESS srsUE started with pid 21546
 * 09:14:40 default - CRITICAL You pressed Ctrl+C!
 * 09:14:40 default - ERROR Exiting...
```

A look at the srslte.backtrace.crash below.

```
--- command='./srsLTE/build/lib/examples/cell_measurement -z 8763' version=18.12.0 signal=13 date='20/08/2020 09:13:43'$
        ./srsLTE/build/lib/examples/cell_measurement(+0x58128a) [0x55c14971128a]
        /lib/x86_64-linux-gnu/libc.so.6(+0x3efd0) [0x7f54b3583fd0]
        /lib/x86_64-linux-gnu/libc.so.6(__write+0x47) [0x7f54b3655297]
        /lib/x86_64-linux-gnu/libc.so.6(_IO_file_write+0x2d) [0x7f54b35d022d]
        /lib/x86_64-linux-gnu/libc.so.6(_IO_file_xsputn+0x1ff) [0x7f54b35d0b9f]
        /lib/x86_64-linux-gnu/libc.so.6(+0x5e787) [0x7f54b35a3787]
        /lib/x86_64-linux-gnu/libc.so.6(_IO_vfprintf+0x396) [0x7f54b35a07b6]
        /lib/x86_64-linux-gnu/libc.so.6(__fprintf_chk+0x106) [0x7f54b3677676]
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(SoapySDR_vlogf+0x65) [0x7f54b33352a5]
        /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libHackRFSupport.so(+0xadc4) [0x7f54b2ed7dc4]
        /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libHackRFSupport.so(_ZN11SoapyHackRFC2ERKSt3mapINSt7__cxx1112basi$
        /usr/lib/x86_64-linux-gnu/SoapySDR/modules0.7/libHackRFSupport.so(+0x7901) [0x7f54b2ed4901]
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(_ZNSt17_Function_handlerIFSt10unique_ptrINSt13__future_base12_Resu$
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(+0x22379) [0x7f54b330b379]
        /lib/x86_64-linux-gnu/libpthread.so.0(+0xf827) [0x7f54b406c827]
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(_ZNSt13__future_base15_Deferred_stateINSt6thread8_InvokerISt5tuple$
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(+0x22ddf) [0x7f54b330bddf]
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(_ZN8SoapySDR6Device4makeERKSt3mapINSt7__cxx1112basic_stringIcSt11c$
        /usr/lib/x86_64-linux-gnu/libSoapySDR.so.0.7(SoapySDRDevice_make+0x38d) [0x7f54b3334f6d]
        /home/mrclean/crocodilehunter/src/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(rf_soapy_open_multi+0x18c) [0x7f5$
        /home/mrclean/crocodilehunter/src/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(srslte_rf_open_devname+0x61) [0x7$
        ./srsLTE/build/lib/examples/cell_measurement(main+0x118) [0x55c1492c2c98]
        /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7f54b3566b97]
        ./srsLTE/build/lib/examples/cell_measurement(_start+0x2a) [0x55c1492c48da]
```

alphafox02, 2020-08-20:
I’ll have a lime tomorrow or Monday. Also if you look at this link

https://github.com/EFForg/crocodilehunter/issues/84

It shows how we can build work 20.04. Which I’m testing now and hoping to just include in m new DragonOS project.

alphafox02, 2020-08-20:
LimeSDR mini in hand. It looks promising as it starts and runs with soapy but I’m getting some error about insufficient rx gain etc.. I think I need to get the device args right and manually place them in the crocodile file after I figure out what’s right running the cell measurement tool on its own. I’ll video it all tonight and see if anyone has advice, but again it looks promising at least with my quick check on 20.04 with latest LimeSuite from the PPa.

There’s still checking it in 18.04 and with different s versions of soapy etc etc.. hoping it’s just the device args though that’ll make it work regardless of other variables.

alexballas, 2020-08-25:
> I’ve found a way to change the clock reference in the lime suite GUI but I think there’s a difference between the normal LimeSDR and the mini.
> 
> https://discourse.myriadrf.org/t/limesdr-on-srslte-unable-to-find-network-on-lte-device/5182/2

Unfortunately any changes applied to the GUI are relevant while the board is connected to it. Any configs in the GUI will go away as soon as you disconnect the board from it.

alphafox02, 2020-09-10:
Not sure if this would help any, but I’ve tried this program with the LimeSDR mini and it seems to work, https://github.com/falkenber9/falcon

alexballas, 2020-09-12:
I got crocodilehunter working on LimeSDR mini (ubuntu 18.04). I found 2 issues:

1) When crocodilehunter first loads I get the following error
ValueError: invalid literal for int() with base 10: '30 ;number of seconds until we assume process has crashed. On a raspberry pi this needs to be more like 120'. I don't see any impact so far.

2) ValueError: Not a boolean: false ;check if towers are operating with the MCCs and MNCs below, they vary by country. If not sure leave it to false
This error keeps popping when finding towers. I don't see any impact so far.

3) (more important): I've setup my config with 5 earfcns values. cell_measurement sub-process memory keeps going up non-stop. In just two minutes it's ~ 7GiB (res) memory. I keep killing it before it maxes out my laptop.

Strangely, If I try to run cell_measurement seperately, it will run for a while, locate some CELLs and then crash.

`$ cat ./srsLTE.backtrace.crash
--- command='./srsLTE/build/lib/examples/cell_measurement -z ***' version=18.12.0 signal=6 date='12/09/2020 22:17:12' ---
	./srsLTE/build/lib/examples/cell_measurement(+0x57df1a) [0x55f8059d6f1a]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3efd0) [0x7fe4884a1fd0]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7fe4884a1f47]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7fe4884a38b1]
	/lib/x86_64-linux-gnu/libc.so.6(+0x89907) [0x7fe4884ec907]
	/lib/x86_64-linux-gnu/libc.so.6(+0x9097a) [0x7fe4884f397a]
	/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6b4) [0x7fe4884fb074]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x903d) [0x7fe4878e703d]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0xf0) [0x7fe4878e8130]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_completed+0x30) [0x7fe4878e8280]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xab99) [0x7fe4878e8b99]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xac9d) [0x7fe4878e8c9d]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_bulk_transfer+0xf) [0x7fe4878e903f]
	/usr/local/lib/libLimeSuite.so.20.07-1(+0x9046f) [0x7fe487b8646f]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime14LMS64CProtocol14TransferPacketERNS0_13GenericPacketE+0x167) [0x7fe487b49cd7]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime14LMS64CProtocol15ReadLMS7002MSPIEPKjPjmj+0x2c8) [0x7fe487b45588]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M14SPI_read_batchEPKtPtt+0x1eb) [0x7fe487b30c7b]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M8SPI_readEtbPi+0xcc) [0x7fe487b30fec]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M19Modify_SPI_Reg_bitsEthhtb+0x2d) [0x7fe487b31e2d]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M11EnableSXTDDEb+0x1e) [0x7fe487b333be]
	/usr/local/lib/libLimeSuite.so.20.07-1(+0x74dd5) [0x7fe487b6add5]
	/usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so(+0xbd68) [0x7fe487dd7d68]
	/usr/local/lib/libSoapySDR.so.0.7(SoapySDRDevice_setFrequency+0x420) [0x7fe488250140]
	/home/alex/test/crocodilehunter/src/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(rf_soapy_set_rx_freq+0x16) [0x7fe4897c3356]
	./srsLTE/build/lib/examples/cell_measurement(main+0x3d9) [0x55f80558bf69]
	/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7fe488484b97]
	./srsLTE/build/lib/examples/cell_measurement(_start+0x2a) [0x55f80558d8ca]

--- command='./srsLTE/build/lib/examples/cell_measurement -z ***' version=18.12.0 signal=6 date='12/09/2020 22:23:07' ---
	./srsLTE/build/lib/examples/cell_measurement(+0x57df1a) [0x55a4bdc0bf1a]
	/lib/x86_64-linux-gnu/libc.so.6(+0x3efd0) [0x7fa1a6894fd0]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7) [0x7fa1a6894f47]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x141) [0x7fa1a68968b1]
	/lib/x86_64-linux-gnu/libc.so.6(+0x89907) [0x7fa1a68df907]
	/lib/x86_64-linux-gnu/libc.so.6(+0x9097a) [0x7fa1a68e697a]
	/lib/x86_64-linux-gnu/libc.so.6(cfree+0x6b4) [0x7fa1a68ee074]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0x903d) [0x7fa1a5cda03d]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_timeout_completed+0xf0) [0x7fa1a5cdb130]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_handle_events_completed+0x30) [0x7fa1a5cdb280]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xab99) [0x7fa1a5cdbb99]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(+0xac9d) [0x7fa1a5cdbc9d]
	/lib/x86_64-linux-gnu/libusb-1.0.so.0(libusb_bulk_transfer+0xf) [0x7fa1a5cdc03f]
	/usr/local/lib/libLimeSuite.so.20.07-1(+0x9046f) [0x7fa1a5f7946f]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime14LMS64CProtocol14TransferPacketERNS0_13GenericPacketE+0x167) [0x7fa1a5f3ccd7]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime14LMS64CProtocol15ReadLMS7002MSPIEPKjPjmj+0x2c8) [0x7fa1a5f38588]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M14SPI_read_batchEPKtPtt+0x1eb) [0x7fa1a5f23c7b]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M8SPI_readEtbPi+0xcc) [0x7fa1a5f23fec]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M19Modify_SPI_Reg_bitsEthhtb+0x2d) [0x7fa1a5f24e2d]
	/usr/local/lib/libLimeSuite.so.20.07-1(_ZN4lime8LMS7002M11EnableSXTDDEb+0x1e) [0x7fa1a5f263be]
	/usr/local/lib/libLimeSuite.so.20.07-1(+0x74dd5) [0x7fa1a5f5ddd5]
	/usr/local/lib/SoapySDR/modules0.7/libLMS7Support.so(+0xbd68) [0x7fa1a61cad68]
	/usr/local/lib/libSoapySDR.so.0.7(SoapySDRDevice_setFrequency+0x420) [0x7fa1a6643140]
	/home/alex/test/crocodilehunter/src/srsLTE/build/lib/src/phy/rf/libsrslte_rf.so(rf_soapy_set_rx_freq+0x16) [0x7fa1a7bb6356]
	./srsLTE/build/lib/examples/cell_measurement(main+0x3d9) [0x55a4bd7c0f69]
	/lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7) [0x7fa1a6877b97]
	./srsLTE/build/lib/examples/cell_measurement(_start+0x2a) [0x55a4bd7c28ca]
`

alphafox02, 2020-09-12:
You’re first few errors are due to the config file having comments. Remove the ; and comment following it.

I ended up trying the mini again and found that it located one cell when I let it run using uhd soapy vs forcing it to just the straight normal soapy. I would always get an error mx code 4 and something about the tx Calibration which is mentioned a lot on forums. Considering you found cells and I found one I have some hope it’ll be usable with some more work.

alexballas, 2020-09-12:
Yes, I also get the
```
[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
```
but only once.

I fixed the first errors as you said.
Would it be possible to check the memory usage of cell_measurement?

cooperq, 2020-09-15:
>  Would it be possible to check the memory usage of cell_measurement?

The memory leak error was fixed in the commit referenced in #84. If you pull the latest release (and update the srsLTE sub repo) you will get the fix.

cooperq, 2020-09-15:
I just ran cell_measurement on my machine with a lime SDR mini and also get the error above:
`[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)`
cell measurement goes as far as decoding the mib on the first pass but can't seem to decode it on the second pass or decode the sib. Going to look into this more. Once we get it working with the C code I will fix the bash script. 

```
$ ./lib/examples/cell_measurement -z 5035,5230,5110,5330,5780
- Scanning 5 EARFCNs
Opening RF device...
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Opening USRP with args: 
Error opening UHD: code 11
Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 2.0] 1D3CCBDE66D3FA, media=USB 2.0, module=FT601, name=LimeSDR Mini, serial=1D3CCBDE66D3FA, 
[INFO] Make connection: 'LimeSDR Mini [USB 2.0] 1D3CCBDE66D3FA'
[INFO] Reference clock 40.00 MHz
[INFO] Device name: LimeSDR-Mini
[INFO] Reference: 40 MHz
[INFO] LMS7002M register cache: Disabled
[INFO] RX LPF configured
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
[INFO] TX LPF configured
Set Tx bandwidth to 5.00 MHz
Setting up RX stream
Setting up TX stream
Available device sensors: 
 - clock_locked
 - lms7_temp
Available sensors for RX channel 0: 
 - lo_locked
[ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
[INFO] Rx calibration finished
State of gain elements for Rx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
State of gain elements for Tx channel 0 (AGC not supported):
 - TIA: 9.00 dB
 - LNA: 30.00 dB
 - PGA: -4.00 dB
Rx antenna set to LNAH
Tx antenna set to BAND1
Starting AGC thread...
Set master clock rate to 245.76 MHz
******* first init multi
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
******* first init multi done 
[INFO] Selected RX path: LNAW
Tuned Rx to 731.50 MHz
[  0/5]: EARFCN 5035 Freq. 731.50 MHz looking for PSS.
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
Tuned Rx to 751.00 MHz
[  1/5]: EARFCN 5230 Freq. 751.00 MHz looking for PSS.
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
Tuned Rx to 739.00 MHz
[  2/5]: EARFCN 5110 Freq. 739.00 MHz looking for PSS.
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
******* rf_mib_decoder
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
Set Rx bandwidth to 2.50 MHz
[INFO] Rx calibration finished
******* rf_mib_decoder_done
Found CELL ID 416. 50 PRB, 2 ports
Tunning receiver to 739.000 MHz
Set master clock rate to 245.76 MHz
Setting sampling rate 11.52 MHz
[INFO] RX LPF configured
Set Rx bandwidth to 8.64 MHz
******* init multi
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
***** plan dft 1d
******* start rx stream
[INFO] Rx calibration finished
******* start rx stream done
******* Begin SIB Decoding Loop******* Decoding MIB try 1
******* Decoding MIB try 2
******* Decoding MIB try 3
******* Decoding MIB try 4
******* Decoding MIB try 5
******* Decoding MIB try 6
******* Decoding MIB try 7
******* Decoding MIB try 8
******* Decoding MIB try 9
******* Decoding MIB try 10
******* Decoding MIB try 11
******* Decoding MIB try 12
******* Decoding MIB try 13
******* Decoding MIB try 14
******* Decoding MIB try 15
******* Decoding MIB try 16
******* Decoding MIB try 17
******* Decoding MIB try 18
******* Decoding MIB try 19
******* Decoding MIB try 20
******* Decoding MIB try 21
Tuned Rx to 763.00 MHz    7.8, FrameCnt: 0, State: 1
[  3/5]: EARFCN 5330 Freq. 763.00 MHz looking for PSS.
[INFO] RX LPF configured
Set Rx bandwidth to 2.50 MHz
[ERROR] Rx calibration: MCU error 4 (SXT tune failed)
```

cooperq, 2020-09-15:
Update: **IT WORKS** 
Apparently the solution was to plug it into a USB3 port and connect some better antennas! Here is the output from SoapySDRUtil:
```
$ SoapySDRUtil --find
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################
Found device 0
  addr = 24607:1027
  driver = lime
  label = LimeSDR Mini [USB 3.0] 1D3CCBDE66D3FA
  media = USB 3.0
  module = FT601
  name = LimeSDR Mini
  serial = 1D3CCBDE66D3FA
```
and here is the output from crocoidle hunter :
```
* 11:10:44 home2 - INFO starting crocodile hunter project: home2
 * 11:10:44 home2 - INFO Starting WebUI
 * 11:10:44 home2 - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 11:10:44 home2 - INFO Checking for config file
 * 11:10:44 home2 - INFO Skipping GPS test
 * 11:10:44 home2 - INFO Compiling srsUE
 * 11:10:44 home2 - INFO SrsUE built succesfully
 * 11:10:44 home2 - INFO Reticulating Splines
 * 11:10:44 home2 - SUCCESS Watchdog server running
 * 11:10:44 home2 - NOTICE Using earfcn list <snip>
 * 11:10:44 home2 - INFO Running srsUE
 * 11:10:44 home2 - SUCCESS srsUE started with pid 120685
 * 11:11:05 home2 - SUCCESS Adding a new tower: <snip>
 * 11:11:05 home2 - INFO Calculating suspiciousness for <snip>
 * 11:11:05 home2 - VERBOSE Found 1 towers a total of 1 times
```

And it goes on like that finding several other cells. The only change I made was to remove the SDR checking stuff from the bash script. 0ab5ac52635569912b8d474f9d213d7c61ab141f In conclusion try the following steps: 
1. Plug into usb3 
2. Pull the latest release 
3. Get an LTE antenna or get up high if you are using the stubby antennas that come with the limeSDR mini. 

If those don't let you run crocodile hunter with the limeSDR mini please comment here.

alexballas, 2020-09-15:
Great work @cooperq !
I just tested with the latest version and cell_measurement seems to be doing find in terms of memory usage so far. Will test more.

I had mine running for 5 minutes (indoors) and got
->   * 21:27:19 default - VERBOSE Found 4 towers a total of 27 times

I had one last question about the towers I get.

I seem to be always getting a "eNodeb id" with value 0 and a PLMN value of 0-0. I get other ledgit towers, but that entry is always there too.

cooperq, 2020-09-15:
yea I've noticed that in the past. I've been told that a 0-0-0 cell can be broadcast when an eNodeB is rebooting sometimes. You should go track it down using the GPS and direction finding!

alphafox02, 2020-09-15:
This is great news. Thanks for making so many improvements. I’ll test it out ASAP. I’ve been using usb3 and good antennas so hopefully the improvements in the code will yield good results. A mini is such a great priced radio to use for this project!

alexballas, 2020-09-15:
On another subject, MyriadRF confirmed [here](https://discourse.myriadrf.org/t/me-and-my-limesdr-usb/6525/4) that they're working on a native version of the LimeSDR library for srsLTE which will also (hopefully) make things much smoother.

alphafox02, 2020-09-15:
Speaking of uhd and soapy.. Are you guys now testing and finding good results with just soapy or uhd soapy when running the crocodile hunter?

cooperq, 2020-09-16:
I was using just soapy with the lime rf. I'll try UHD Soapy as well with the UHD device I have though.

alphafox02, 2020-09-16:
I just mentioned uhd soapy because if I build this and have the uhd driver and soapy installed uhd Soapy takes precedence with the lime sdr until I go disable uhd in the cmake file and build srslte with just soapy support.

cooperq, 2020-09-16:
does this cause a bug? Can you show me what the output is? I'm a little slow today but I don't totally understand the problem.

alphafox02, 2020-09-16:
Built the latest and it works great so far with the lime-sdr. I didn’t change a thing at first so when I started it up it was using the “uhd soapy” with the LimeSDR. It seemed to run but something didn’t seem right, didn’t run it long enough to really evaluate it well. 

So I stopped it, went in to srsLTE and changed the CMakefile list file to disable uhd and enable soapy. Rebuilt and reran and within a few minutes the LimeSDR found two cells/towers. It looked like it was running way faster when compared to uhd soapy. 

Thanks for making this possible!

alphafox02, 2020-09-16:
This is with the CMakelist file like this, which I had to manually do and it seems to run faster with the limesdr.

option(ENABLE_GUI      "Enable GUI (using srsGUI)"                OFF)
option(ENABLE_UHD      "Enable UHD"                               OFF)
option(ENABLE_BLADERF  "Enable BladeRF"                           ON)
option(ENABLE_SOAPYSDR "Enable SoapySDR"                          ON)

 * 13:56:43 default - INFO Running srsUE
 * 13:56:43 default - SUCCESS srsUE started with pid 11187
 * 13:56:44 default - DEBUG [INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
 * 13:56:44 default - DEBUG [INFO] Make connection: 'LimeSDR Mini [USB 3.0] 1D4C3CEC7D3A8A'
 * 13:56:44 default - DEBUG [INFO] Reference clock 40.00 MHz
 * 13:56:44 default - DEBUG [INFO] Device name: LimeSDR-Mini
 * 13:56:44 default - DEBUG [INFO] Reference: 40 MHz
 * 13:56:44 default - DEBUG [INFO] LMS7002M register cache: Disabled
 * 13:56:47 default - DEBUG [INFO] RX LPF configured
 * 13:56:48 default - DEBUG [INFO] RX LPF configured
 * 13:56:48 default - DEBUG [INFO] Filter calibrated. Filter order-4th, filter bandwidth set to 5 MHz.Real pole 1st order filter set to 2.5 MHz. Preemphasis filter not active
 * 13:56:48 default - DEBUG [INFO] TX LPF configured
 * 13:56:48 default - DEBUG [ERROR] Tx Calibration: MCU error 4 (SXT tune failed)
 * 13:56:48 default - DEBUG [INFO] Rx calibration finished
 * 13:56:48 default - DEBUG Invalid DL_EARFCN=68686
 * 13:56:48 default - DEBUG ******* first init multi
 * 13:56:48 default - DEBUG ***** plan dft 1d
 * 13:56:48 default - DEBUG ***** plan dft 1d
 * 13:56:48 default - DEBUG ***** plan dft 1d
 * 13:56:48 default - DEBUG ***** plan dft 1d
 * 13:56:48 default - DEBUG ***** plan dft 1d
 * 13:56:48 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ***** plan dft 1d
 * 13:56:49 default - DEBUG ******* first init multi done
 * 13:56:49 default - DEBUG - Scanning 38 EARFCNs
 * 13:56:49 default - DEBUG Opening RF device...
 * 13:56:49 default - DEBUG Soapy has found device #0: addr=24607:1027, driver=lime, label=LimeSDR Mini [USB 3.0] 1D4C3CEC7D3A8A, media=USB 3.0, module=FT601, name=LimeSDR Mini, serial=1D4C3CEC7D3A8A,                                                                                                                                                                                  
 * 13:56:49 default - DEBUG Set Rx bandwidth to 2.50 MHz
 * 13:56:49 default - DEBUG Set Tx bandwidth to 5.00 MHz
 * 13:56:49 default - DEBUG Setting up RX stream
 * 13:56:49 default - DEBUG Setting up TX stream
 * 13:56:49 default - DEBUG Available device sensors:
 * 13:56:49 default - DEBUG  - clock_locked
 * 13:56:49 default - DEBUG  - lms7_temp
 * 13:56:49 default - DEBUG Available sensors for RX channel 0:
 * 13:56:49 default - DEBUG  - lo_locked
 * 13:56:49 default - DEBUG State of gain elements for Rx channel 0 (AGC not supported):
 * 13:56:49 default - DEBUG  - TIA: 9.00 dB
 * 13:56:49 default - DEBUG  - LNA: 30.00 dB
 * 13:56:49 default - DEBUG  - PGA: -4.00 dB
 * 13:56:49 default - DEBUG State of gain elements for Tx channel 0 (AGC not supported):
 * 13:56:49 default - DEBUG  - TIA: 9.00 dB
 * 13:56:49 default - DEBUG  - LNA: 30.00 dB
 * 13:56:49 default - DEBUG  - PGA: -4.00 dB
 * 13:56:49 default - DEBUG Rx antenna set to LNAH
 * 13:56:49 default - DEBUG Tx antenna set to BAND1
[... 69 lines cut ...]
 * 14:02:53 default - DEBUG ***** plan dft 1d
 * 14:02:53 default - DEBUG ***** plan dft 1d
 * 14:02:53 default - DEBUG ***** plan dft 1d
 * 14:02:53 default - DEBUG ***** plan dft 1d
 * 14:02:53 default - DEBUG ******* first init multi done
 * 14:02:53 default - DEBUG - Scanning 38 EARFCNs
 * 14:02:53 default - DEBUG Opening RF device...
 * 14:02:53 default - DEBUG Opening USRP with args:
 * 14:02:53 default - DEBUG Starting AGC thread...
 * 14:02:53 default - DEBUG [  0/38]: EARFCN 2050 Freq. 2120.00 MHz looking for PSS.
 * 14:02:53 default - DEBUG [FATAL] [UHDSoapyDevice] Tx Calibration: MCU error 4 (SXT tune failed)
 * 14:02:53 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
 * 14:02:54 default - DEBUG [  1/38]: EARFCN 900 Freq. 1960.00 MHz looking for PSS.
 * 14:02:54 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
 * 14:02:54 default - DEBUG [  2/38]: EARFCN 41094 Freq. 3073.50 MHz looking for PSS.
 * 14:02:55 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
 * 14:02:55 default - DEBUG [  3/38]: EARFCN 650 Freq. 1935.00 MHz looking for PSS.
 * 14:02:55 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
 * 14:02:56 default - DEBUG [  4/38]: EARFCN 41490 Freq. 3113.10 MHz looking for PSS.
 * 14:02:56 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
 * 14:02:56 default - DEBUG [  5/38]: EARFCN 40978 Freq. 3061.90 MHz looking for PSS.
 * 14:02:57 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
 * 14:02:57 default - DEBUG [  6/38]: EARFCN 66836 Freq. 2150.00 MHz looking for PSS.
 * 14:02:57 default - DEBUG [INFO] [UHDSoapyDevice] Rx calibration finished
^ * 14:02:57 default - CRITICAL You pressed Ctrl+C!

[5 more replies]

closed 2020-10-15 by cooperq