EFForg/crocodilehunter #89: Use outside of the US

marcfielding1, 2020-08-10:
Heya,

A great piece of work on this! I'm based just north of London and would be interested in testing this, Stingray usage in and around London has been shown to be pretty prevalent both via law enforcement and criminals.

When I watched the talk on this I noticed console logs talking about it giving false positives outside the US, would I be able to reconfigure Crocodile Hunter to work correctly here?

I would say this isn't my area of expertise, however, once I've had a little time to study it I'm sure I'll get it down, also the company I own does have some pretty nifty machine learning resource so once I understand it I'd look to apply those particular skills to the data.

Thanks

Marc

cooperq, 2020-08-14:
you need to set `run_us_centeric_heuristics = false` in src/config.ini 
other than that it should work just fine in london. Good luck!

cooperq, 2020-08-25:
actually the latest pull request added a new config flag where you can set the country codes for your own area. I would reccomend using this feature!

closed 2020-08-25 by cooperq