EFForg/crocodilehunter #90: MemoryError

keldnorman, 2020-08-11:
root@gateway:~/source/4G_Crocodilehunter# cat /etc/*release
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2020.3"
VERSION_ID="2020.3"
VERSION_CODENAME="kali-rolling"

- Clean install of Kali on a VM
- Followed the install guide 
- Filesystem not full
- Memory allocated: 32 Gb
- CPU: 2
- GPU memory 3Gb and support for acceleration ticked on in the vmware profile.

when starting it complains a little about some error in the code related to fabsf ? but it starts up.

 * 10:33:01 dubex - INFO cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
 * 10:33:01 dubex - INFO /root/source/4G_Crocodilehunter/crocodilehunter/src/srsLTE/srsue/src/phy/phch_worker.cc:49:2: warning: #warning Compiling without srsGUI support [-Wcpp]
 * 10:33:01 dubex - INFO    49 | #warning Compiling without srsGUI support
 * 10:33:01 dubex - INFO       |  ^~~~~~~
 * 10:33:02 dubex - INFO /root/source/4G_Crocodilehunter/crocodilehunter/src/srsLTE/srsue/src/phy/phch_worker.cc: In member function ‘int srsue::phch_worker::read_ce_abs(float*, uint32_t, uint32_t)’:
 * 10:33:02 dubex - INFO /root/source/4G_Crocodilehunter/crocodilehunter/src/srsLTE/srsue/src/phy/phch_worker.cc:1599:31: error: ‘cabsf’ was not declared in this scope; did you mean ‘fabsf’?
 * 10:33:02 dubex - INFO  1599 |     ce_abs[g+i] = 20 * log10f(cabsf(ue_dl.ce_m[tx_antenna][rx_antenna][i]));
 * 10:33:02 dubex - INFO       |                               ^~~~~
 * 10:33:02 dubex - INFO       |                               fabsf
 * 10:33:02 dubex - INFO make[2]: *** [srsue/src/phy/CMakeFiles/srsue_phy.dir/build.make:89: srsue/src/phy/CMakeFiles/srsue_phy.dir/phch_worker.cc.o] Error 1
 * 10:33:02 dubex - INFO make[1]: *** [CMakeFiles/Makefile2:4923: srsue/src/phy/CMakeFiles/srsue_phy.dir/all] Error 2
 * 10:33:02 dubex - INFO make: *** [Makefile:163: all] Error 2
 * 10:33:02 dubex - INFO SrsUE built succesfully
* 10:33:02 dubex - INFO Reticulating Splines
 * 10:33:02 dubex - SUCCESS Watchdog server running
 * 10:33:02 dubex - NOTICE Using earfcn list 2850,199,1644,37900,6350,174,3150,1650,1300,500,3348,3000,6200,1850,9435,1500,1471
 * 10:33:02 dubex - INFO EARFCNS: 2850,199,1644,37900,6350,174,3150,1650,1300,500,3348,3000,6200,1850,9435,1500,1471
 * 10:33:02 dubex - INFO Running srsUE
 * 10:33:02 dubex - SUCCESS srsUE started with pid 54720

Then it runs for some minutes and then dies with Memory error - chrome or firefox also stops working. 
I have given it 32 Gb memory in the VM and it does not look like it has used the memory ( i base this on visual monitoring the processes with the utility nmon while running crocodilehunter )

----------------------------------------
 * 10:37:44 dubex - SUCCESS Adding a new tower: <Tower: 238-1-2800-63262.0, loc: 55.7261813,12.477495215, time: 2020-08-11 10:37:43, freq: 780.5>
 * 10:37:44 dubex - INFO Calculating suspiciousness for <Tower: 238-1-2800-63262, loc: 55.7262,12.4775, time: 2020-08-11 10:37:43, freq: 780.5>
----------------------------------------
Exception happened during processing of request from 
Traceback (most recent call last):
  File "/usr/lib/python3.8/socketserver.py", line 316, in _handle_request_noblock
    self.process_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 347, in process_request
    self.finish_request(request, client_address)
  File "/usr/lib/python3.8/socketserver.py", line 360, in finish_request
    self.RequestHandlerClass(request, client_address, self)
  File "/usr/lib/python3.8/socketserver.py", line 720, in __init__
    self.handle()
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/watchdog.py", line 536, in handle
    wd_inst.process_tower(data)
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/watchdog.py", line 267, in process_tower
    self.calculate_suspiciousness(new_tower)
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/watchdog.py", line 472, in calculate_suspiciousness
    self.check_wigle(tower)
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/watchdog.py", line 419, in check_wigle
    resp = self.wigle.cell_search(tower.lat, tower.lon, 0.1, tower.cid, tower.tac)
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/wigle.py", line 79, in cell_search
    return self._api_request("cell/search", params)
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/wigle.py", line 42, in _api_request
    while prev_resp['searchAfter'] is not None:
KeyError: 'searchAfter'
----------------------------------------
Exception in thread Thread-4:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 870, in run
  File "/root/source/4G_Crocodilehunter/crocodilehunter/src/nbstreamreader.py", line 24, in _populateQueue

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 874, in run
MemoryError
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "./crocodilehunter.py", line 156, in monitor_srslte
    out.append(line)
MemoryError

Is this related to the error when starting up and not being able to compile phch_worker.cc ?

keldnorman, 2020-08-11:
Found this link that fixex the compile error with phch_worker.cc.:  https://git.mmcginty.me/mirror/srsLTE_srsLTE/commit/aa3a2a3dba563540d98987a20d8de34389291ebc.diff

Just alter line 1599 from this: 

ce_abs[g+i] = 20 * log10f(cabsf(ue_dl.ce_m[tx_antenna][rx_antenna][i]));

to this: 

ce_abs[g+i] = 20 * log10f(std::abs(std::complex<float>(ue_dl.ce_m[tx_antenna][rx_antenna][i])));

cooperq, 2020-08-14:
nice find! I'll fix that!

cguerrave, 2020-08-15:
with this exact fix I still had problems at some point of the build, basing on that same diff @keldnorman  I changed the line of interest to this slightly different version as pointed in the diff file:

``` ce_abs[g+i] = 20 * log10f(std::abs(std::complex<float>(ue_dl.ce_m[tx_antenna][rx_antenna][i]))); ```

And the build was made without any issue.

cguerrave, 2020-08-18:
> ----------------------------------------
> Exception happened during processing of request from
> Traceback (most recent call last):
>   File "/usr/lib/python3.8/socketserver.py", line 316, in [...]

With the fix, the cabsf error disappears but this error above keeps showing, maybe is caused by another reason

cooperq, 2020-08-18:
yea that is a separate python error with my wigle API code.

cooperq, 2020-09-15:
fixed in https://github.com/EFForg/srsLTE/commit/2799232026df9f466703a0f9870923b44d902270

closed 2020-09-15 by cooperq