EFForg/crocodilehunter #91: HackRF

PowerPress, 2020-08-12:
Any chance of HackRF hardware support to increase userbase since it is much cheaper hardware?

enrutador, 2020-08-12:
YES ¡¡¡¡ I also support the opinion of PowerPress give support to hackrf. If we put a crowdfunding code, or address of paypal I would collaborate.
Thanks for your work

pdelteil, 2020-08-12:
I don't think it will work properly. The HackRF is a half-duplex device with a much lower throughput than the BladeRF, but I'm not sure.

cooperq, 2020-08-14:
see #86 and follow the same steps. Since we don't do any transmission a hackRF might work but as @pdelteil mentions its half duplex so I would be surprised. Good luck!

cooperq, 2020-08-17:
Closing as a duplicate of #86. Please follow up there.

closed 2020-08-17 by cooperq