EFForg/crocodilehunter #93: Changing the run_us_centeric_heuristics approach to "expected MXC"

cguerrave, 2020-08-16:
Knowing the mccs and mncs of the places we want to monitor, maybe we can have a couple of lines in config.ini with:
```
expected_mccs = 123,456,789
expected_mncs = 2,4,6,102
``` 
And maybe change the run_us_centeric_heuristics to ```check_mcc_mnc``` or something like that (?)

cooperq, 2020-08-17:
Yea this seems like a better way to do it. I still don't understand why we ever see this and I don't feel like it should actually be a huristic but I've seen it myself in the wild so I guess we might as well keep doing it?

cguerrave, 2020-08-17:
Perfect! also in the monitoring in 2G made so fat the weirdest towers had weird MCCs and MNCs so this seems nice to have in Crocodile Hunter too outside the US, I'm working on this in a branch in my fork, I'll let you know how it went (still learning a lot with this repo and git :) )

cooperq, 2020-08-25:
fixed in #95

closed 2020-08-25 by cooperq