EFForg/crocodilehunter #97: Unable to run...

Al-gore-ithm, 2020-08-30:
Hi,

I'd be grateful for a little assistance, I am trying to run Crocodile Hunter without success so far. I have successfully built and tested srsLTE, but I've no idea what I need to do resolve this. The output is below, any and all suggestions are welcome...

* 10:43:09 default - INFO starting crocodile hunter project: default
 * 10:43:09 default - INFO Starting WebUI
 * 10:43:09 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 10:43:09 default - INFO Checking for config file
 * 10:43:09 default - INFO Checking for SDR
 * 10:43:09 default - INFO BladeRF Found
 * 10:43:10 default - INFO Skipping GPS test
 * 10:43:10 default - INFO Compiling srsUE
 * 10:43:13 default - INFO cc1plus: warning: ‘-Werror=’ argument ‘-Werror=incompatible-pointer-types’ is not valid for C++
 * 10:43:13 default - INFO /home/al/crocodilehunter/src/srsLTE/srsue/src/phy/phch_worker.cc:49:2: warning: #warning Compiling without srsGUI support [-Wcpp]
 * 10:43:13 default - INFO    49 | #warning Compiling without srsGUI support
 * 10:43:13 default - INFO       |  ^~~~~~~
 * 10:43:15 default - INFO /home/al/crocodilehunter/src/srsLTE/srsue/src/phy/phch_worker.cc: In member function ‘int srsue::phch_worker::read_ce_abs(float*, uint32_t, uint32_t)’:
 * 10:43:15 default - INFO /home/al/crocodilehunter/src/srsLTE/srsue/src/phy/phch_worker.cc:1599:31: error: ‘cabsf’ was not declared in this scope; did you mean ‘fabsf’?
 * 10:43:15 default - INFO  1599 |     ce_abs[g+i] = 20 * log10f(cabsf(ue_dl.ce_m[tx_antenna][rx_antenna][i]));
 * 10:43:15 default - INFO       |                               ^~~~~
 * 10:43:15 default - INFO       |                               fabsf
 * 10:43:15 default - INFO make[2]: *** [srsue/src/phy/CMakeFiles/srsue_phy.dir/build.make:89: srsue/src/phy/CMakeFiles/srsue_phy.dir/phch_worker.cc.o] Error 1
 * 10:43:15 default - INFO make[1]: *** [CMakeFiles/Makefile2:4923: srsue/src/phy/CMakeFiles/srsue_phy.dir/all] Error 2
 * 10:43:15 default - INFO make: *** [Makefile:163: all] Error 2
 * 10:43:15 default - INFO SrsUE built succesfully
 * 10:43:15 default - INFO Reticulating Splines
 * 10:43:15 default - SUCCESS Watchdog server running
 * 10:43:15 default - WARNING Getting earcn list for the first time, this might take a while
 * 10:43:16 default - INFO opencellid location {'status': 'error', 'message': 'Invalid token - get a trial token by signing up for free here: http://unwiredlabs.com/trial', 'balance': 0}
Traceback (most recent call last):
  File "./crocodilehunter.py", line 226, in <module>
    crocodile_hunter.start()
  File "./crocodilehunter.py", line 109, in start
    self.update_earfcn_list()
  File "./crocodilehunter.py", line 131, in update_earfcn_list
    self.earfcn_list = self.watchdog.wigle.earfcn_search(gps.lat, gps.lon, 0.2)
  File "/home/al/crocodilehunter/src/wigle.py", line 96, in earfcn_search
    return set(filter(_noneorzero, [x['channel'] for x in res['results']] ))
TypeError: 'Response' object is not subscriptable

Thanks in advance

Al

cooperq, 2020-09-15:
looks like the problem here is your connection to wigle. Did you get a wigle api key and put it in config.ini? Crocodile hunter needs to connect to Wigle to get a list of EARFCNs (frequencys) to scan. If you don't want to or can't do that you can manually search for the frequencies around you using the wigle or open cell id web interface and put them in the config.ini with a line like so
```
[project-name]
earfcns = 1025,2050,66561,40072,17,40978
```
**NOTE**: make sure there are no spaces between the commas.

Al-gore-ithm, 2020-09-15:
Thank you very much, I've tried that using a Wigle API username and key. I've also tried combinations of standard username and password with API username and API key all with no success. Using EARFCN list I do have some success running but I'm still seeing errors:

_**Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "./crocodilehunter.py", line 154, in monitor_srslte
    line = nbsr.readline(int(self.config["general"]["crash_timeout"]))
ValueError: invalid literal for int() with base 10: '30 ;number of seconds until we assume process has crashed. On a raspberry pi this needs to be more like 120'_**

I'm happy attempting to correct the error in ./crocodilehunter.py line 54, the others are a mystery to me though. Any further suggestions?

I am trying to use a VM running Ubuntu 20.04, srsUE runs ok on its own but this is beating me.

Again, thanks.

cooperq, 2020-09-16:
ah yes, you need to remove the comment (everything after and including the ;) in config.ini  it should parse this but it's a bug I need to fix. after that it should work.

closed 2020-10-15 by cooperq