EFForg/crocodilehunter #101: Unable to run, getting stuck here

Mutekeri, 2020-09-27:
Traceback (most recent call last):
  File "./crocodilehunter.py", line 225, in <module>
    crocodile_hunter.start()
  File "./crocodilehunter.py", line 108, in start
    self.update_earfcn_list()
  File "./crocodilehunter.py", line 130, in update_earfcn_list
    self.earfcn_list = self.watchdog.wigle.earfcn_search(gps.lat, gps.lon, 0.2)
  File "/usr/src/crocodilehunter/src/wigle.py", line 91, in earfcn_search
    res = self._api_request("cell/search", qs_params)
  File "/usr/src/crocodilehunter/src/wigle.py", line 42, in _api_request
    while prev_resp['searchAfter'] is not None:
KeyError: 'searchAfter'

cooperq, 2020-09-28:
do you have a wigle_name and wigle_key entered in your config.ini file?

Mutekeri, 2020-09-29:
Yes I have the wigle_name, wigle_key and open cell ID. I even have the mccs and mncs set up. I am using the crocodile hunter in South Africa, I had done all the configurations, except where you have to update the known towers. I had previously tried to update them using the information, I downloaded from wingle for towers in South Africa, but still I as unable to run. Does the solution only work in US. Maybe if there are clear instructions to update the known towers, it would work.

cooperq, 2020-10-01:
updating the known towers shouldn't be a requirement for crocodile hunter but perhaps you mean known_earfcns not known_towers? I will look at what is going on with the wigle scirpt as that is a bug I need to fix. But the other thing that would help here is if you can paste your config.ini (removing the API keys first). Thanks!

martinmathisen, 2020-10-03:
I am getting more or less the same: 
Traceback (most recent call last):
  File "./crocodilehunter.py", line 226, in <module>
    crocodile_hunter.start()
  File "./crocodilehunter.py", line 109, in start
    self.update_earfcn_list()
  File "./crocodilehunter.py", line 131, in update_earfcn_list
    self.earfcn_list = self.watchdog.wigle.earfcn_search(gps.lat, gps.lon, 0.2)
  File "/home/ubuntu/crocodilehunter/src/wigle.py", line 91, in earfcn_search
    res = self._api_request("cell/search", qs_params)
  File "/home/ubuntu/crocodilehunter/src/wigle.py", line 42, in _api_request
    while prev_resp['searchAfter'] is not None:
KeyError: 'searchAfter'

I have attached my config file
[image]

Mutekeri, 2020-10-07:
[general]
gps_default = 0, 0
mysql_path = mysql://root:toor@localhost:3306
wigle_name = XXXX
wigle_key = XXXX
ocid_key = XXXXXXX
default_project = default
crash_timeout = 130
check_geographic_codes = false
expected_mccs = 655
expected_mncs = 1,2,25,33,11,41,46,6,19,16,14,13,10,53,35,31,51,32,28,50,7,21,34,30,36

[api]
host = 0.0.0.0
port = 5000
contact = test@example.com

[default]

cooperq, 2020-10-14:
From your email: 
> 1. I have formatted my drive to start a new installation of Linux 20.4
>2. I followed all the steps to install blade RF X4
> 3. Installed the srsLTE
> 4. Download the updated repository with Crocodile hunter, where I copied the install srsLTE and deleted the empty folder in crocodile hunter.
> 5. Python version is 3.8.3
> 6. Installed all the dependencies and corrected the error file
> 7. Installed the requirements
> 8. Also ran the docker-compose, also installing other requirements to make
> it run
> 9. Fixed the msql error
> 10. Set up the config file as above
> 11. Ran the ./crocodilehunter.py -d 
> 12. The results are all on the screenshots
> 13. The only thing I did not update is mysql database which you indicated,
> it should work without known towers. The only available data I could get
> access to was from opencellID, for south Africa. Even though I did not
> update it.
> 14. I am using the BU-353S4 for GPS receiver, which I configured in the
> /etc/default/gpsd using nano, adding on  DEVICES="/dev/ttyUSB0"

Interesting it sounds like it's scanning then but not finding anything. I would be very interested to see what the output is when you run it in debug mode with the -d flag like so
`./crocodilehunter.py -d -p <project>`

15. I ran the crocodile overnight, It won't show anything, it just get stick
there

Mutekeri, 2020-10-19:
Thank you very much. Its running fine, without wingle. I used the earfns using Netmonitor. Thank you. I also load DragonOS the latest, on its own machine and updated the config file.

closed 2020-10-21 by cooperq

msfu101, 2021-09-11:
> Thank you very much. Its running fine, without wingle. I used the earfns using Netmonitor. Thank you. I also load DragonOS the latest, on its own machine and updated the config file.

Can you tell me if you use an usrp b210 or another with ext. PLL-GPSDO how you configured your GPS to get the signal from external Clock and set it up for crocodilehunter? 
thanks in advance

commit 36765343 (2020-10-29): switch to new faster API for earfcn finding and fix searchAfter bug in Wigle API. Fixes #101. Fixes #103