EFForg/crocodilehunter #114: (Request) Croc Hunter + Cesium

alphafox02, 2021-02-26:
(Request) Allowing CH database to be read by or feed somehow into cesium. 

Currently using DF Aggregator for direction finding and thought it’d be great to have a layer added to cesium which would plot the cells and maybe allow them to be clicked for expanded info normally shown in the standard mapping GUI. 

https://github.com/ckoval7/df-aggregator

Cesium is installed in DragonOS with a very basic map set. This could add expanded offline capability to CH.

Al3x4n, 2021-03-07:
USRP support?

cooperq, 2021-03-11:
I'm all for this but don't have time to do it personally. I would welcome a pull request though.

alphafox02, 2021-03-11:
One day, when I learn python or how to code, I’ll add this! I think it’d be a cool learning project.

alphafox02, 2021-05-20:
@cooperq what about a tool to output the tower info to a kml instead?

cooperq, 2021-06-10:
that would be cool too but I don't personally have time to do it.

alphafox02, 2021-06-11:
Needs some refinement, but sharing in case you can use it

https://pastebin.com/qq2LWeJa

closed 2021-06-11 by alphafox02