EFForg/crocodilehunter #116: Compile Crocodile Hunter's srsLTE on aarch64

alphafox02, 2021-03-03:
While attempting to buiild Crocodile hunter on a Pi4 running Ubuntu 20.04 aarch64 I find the following is failing, which seems most likely an issue with srsLTE after reading this comment on another program ( https://github.com/F5OEO/rpitx/issues/173 ).

ubuntu@ubuntu:/usr/src/crocodilehunter/src/srsLTE/build$ sudo make -j4
Scanning dependencies of target srslte_asn1
Scanning dependencies of target arch_select
Scanning dependencies of target gen_build_info
[  1%] Building CXX object lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o
[  1%] Building CXX object lib/src/common/CMakeFiles/arch_select.dir/arch_select.cc.o
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfpu=neon’
make[2]: *** [lib/src/asn1/CMakeFiles/rrc_asn1.dir/build.make:63: lib/src/asn1/CMakeFiles/rrc_asn1.dir/rrc_asn1.cc.o] Error 1
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
make[1]: *** [CMakeFiles/Makefile2:2103: lib/src/asn1/CMakeFiles/rrc_asn1.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
c++: error: unrecognized command line option ‘-mfpu=neon’
make[2]: *** [lib/src/common/CMakeFiles/arch_select.dir/build.make:63: lib/src/common/CMakeFiles/arch_select.dir/arch_select.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2157: lib/src/common/CMakeFiles/arch_select.dir/all] Error 2
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
-- Generating build_info.h
[  1%] Built target gen_build_info
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_common.cc.o
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_mme.cc.o
[  2%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_s1ap.cc.o
[  3%] Building CXX object lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_m2ap.cc.o
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfpu=neon’
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfpu=neon’
make[2]: *** [lib/src/asn1/CMakeFiles/srslte_asn1.dir/build.make:63: lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_common.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [lib/src/asn1/CMakeFiles/srslte_asn1.dir/build.make:76: lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_mme.cc.o] Error 1
c++: error: unrecognized command line option ‘-mfpu=neon’
make[2]: *** [lib/src/asn1/CMakeFiles/srslte_asn1.dir/build.make:89: lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_s1ap.cc.o] Error 1
c++: error: unrecognized command line option ‘-mfloat-abi=hard’
c++: error: unrecognized command line option ‘-mfpu=neon’
make[2]: *** [lib/src/asn1/CMakeFiles/srslte_asn1.dir/build.make:102: lib/src/asn1/CMakeFiles/srslte_asn1.dir/liblte_m2ap.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2130: lib/src/asn1/CMakeFiles/srslte_asn1.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

alphafox02, 2021-03-03:
It appears newer builds of srsLTE do work with aarch64 as mentioned here

https://docs.srslte.com/en/latest/app_notes/source/pi4/source/

alphafox02, 2021-03-08:
Aarch support

alphafox02, 2021-03-08:
I got it working by manually applying aarch patches found in later versions of srsLTE. If I figure out the proper way to do a pull request I’ll submit.

cooperq, 2021-03-11:
ah this might be a documentation issue, I have compiled this successfully on a pi4 and it shoudl support arm already. I will look into it.

alphafox02, 2021-03-11:
So I think you’re fine when using a pi image, like raspbian, however if you use the official Ubuntu 20.04 headless image it’s aarch64 and the version of srsLTE you’re using is missing the patches there were added later. I’ll find them again and send you the link. Again for most people this will probably not be noticed as I think the normal pi image does not mark itself as “aarch64”.

alphafox02, 2021-03-11:
I manually added everything here to the version of srsLTE you’re using. 

https://github.com/srsLTE/srsLTE/commit/4a1b8a595261a6f180a9dcb4a8a83a9af3278bf3

cooperq, 2021-03-24:
Nice! Could you open up a pull request for that? I would love to add it to the main branch!

alphafox02, 2021-03-25:
This will be homework for me at some point because I do need to get familiar with doing proper pull requests.