EFForg/crocodilehunter #117: api_client not pushing info into server database

alphafox02, 2021-03-09:
I’ve setup a client crocodile Hunter instance and can successfully appear to signup and create an AP key, followed by pushing towers to the server running on another system, however no data is entered anywhere into the database running on the server

There’s no visible error and I’m not that familiar with the setup yet, but had anyone else ran the server and client setup and had data passed back into the server database?

cooperq, 2021-03-11:
we are running a server with EFF and it has worked. Can you first check if the key is authorized in the database? (you need to manually authorize each api key from the server.)

alphafox02, 2021-03-11:
I think what’s happening is a portion is working, however I did have to go change the client api to http and not https because I was getting ssl or mismatch something errors.. once I switched it to http I was able to register and get a key which I confirmed ended up in the config file of the client. 

However, after I went and command line dug into the MySQL/marinaDB on the server side I quickly found it was not actually entering any information into any of the api tables. The only thing that was happening is that on the server side if I started and used CH it was in fact populating the db, but nothing was getting populated from the client even though it all appeared to be working.

alphafox02, 2021-03-11:
I’ll check the database again, assuming you mean the key is entered into the db on the server side? If that’s the case I recall checking all api tables and really all tables and not seeing the key or anything from the client anywhere.

alphafox02, 2021-03-11:
Oh wait, now I think I see what you’re saying. What do you mean manually authorize each key on the server? When I do the sign up on the client end it generates a key and applies it to the config file. I guess I assumed that key would at the same time be entered in on the server side. That must be the part I’m missing, to somehow manually enter the key on the server side?

alphafox02, 2021-03-12:
I tried it out again. I notice now and maybe I wasn’t looking at it correctly last time, but now I can sign up to the server. I assumed the the CH_project variable would have to do with how the data is saved on the server to recall say the project later on the server with map only.

However, I see after signing up that back on the server I check the database and find an entry called crocodilehunter_api and under this table I can see the api users/api towers etc. I check api users and sure enough there’s the api key for the client already there. I go back to the client and push tower with the ch_project specified as default. 

I come back to the server and can tower see the nine tower entries under the same crocodilehunter_api table within the api towers section. So that appear to work. 

Maybe I’m misunderstanding, but I thought the towers would get saved on the server as the “default” project. Right now that’s not what I’m seeing and if I try to open crocodilehunter_api as the project name I do not see the relevant data I’ve pushed over from the client on the server api map side. Maybe I’m misunderstanding how this all should work.

cooperq, 2021-03-24:
yea no you have it now. It's funny because I'm actually helping someone else out with crocodile hunter now and they are running into all the same bugs. So yea the api data gets sent to a table called api_towers. And there is no good UI to view the data, you just have to view it in mysql.

The entire API was an afterthought that was written in 1 day and I think I need to build a nice dashboard for it and fix some pretty severe bugs. This will be  my top priority in crocodile hunter with work going on here in #119

closed 2021-03-24 by cooperq