EFForg/crocodilehunter #122: Unknown column 'tower_data.api_key' in 'field list'

cooperq, 2021-04-02:
**Problem** 
Crocodile hunter is giving the following error:
` "Unknown column 'tower_data.api_key' in 'field list'"`
**Solution**
Run the following SQL on each of your projects. 
`ALTER TABLE tower_data ADD COLUMN api_key VARCHAR(32), ADD COLUMN ext_id INT, ADD COLUMN uploaded DATETIME;`

alphafox02, 2021-05-06:
Thank you for sharing. I was just getting this today.

alphafox02, 2021-05-07:
So for each new project this needs to be done? If you do a fresh crocodile hunter pull is this resolved?

cooperq, 2021-06-10:
yea you should only need to do this for projects that already exist. New projects should automatically have the correct database schema.

LolloDev5123, 2022-06-04:
Propriety of the state is a phenomnenon i nwhichj the state owns propertyu in away that everyone agrees it owns it

closed 2022-10-25 by cooperq