EFForg/crocodilehunter #128: no gps fix detected... aborting

msfu101, 2021-09-11:
hey, ther
ive got a question about crocodilehunter.
if i do sudo ./crocodilehunter.py -d it says

 * 04:36:11 default - INFO starting crocodile hunter project: default
 * 04:36:11 default - INFO Starting WebUI
 * 04:36:11 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 04:36:11 default - INFO Checking for config file
 * 04:36:11 default - INFO Compiling srsUE
 * 04:36:12 default - INFO SrsUE built succesfully
 * 04:36:12 default - INFO Reticulating Splines
 * 04:36:12 default - DEBUG Connecting to gpsd socket at 127.0.0.1:2947
 * 04:36:12 default - DEBUG Waiting for welcome message
 * 04:36:12 default - DEBUG Enabling gps
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - INFO No GPS fix detected. Waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - DEBUG Try %s waiting for fix.
 * 04:36:12 default - DEBUG Polling gps
 * 04:36:12 default - CRITICAL No GPS fix detected. Exiting.
 * 04:36:12 default - ERROR Unhandled exception:
Traceback (most recent call last):                                                                       
  File "./crocodilehunter.py", line 281, in <module>                                                     
    crocodile_hunter.start()                                                                             
  File "./crocodilehunter.py", line 137, in start                                                        
    exit(1)                                                                                              
  File "/usr/lib/python3.8/_sitebuiltins.py", line 26, in __call__                                       
    raise SystemExit(code)                                                                               
SystemExit: 1                                                                                            
                                                                                                         
 * 04:36:12 default - ERROR Exiting...
 * 04:36:12 default - WARNING Stopping Watchdog
 * 04:36:12 default - SUCCESS See you space cowboy...

so, why i can't get no gps fix? can anybody say what there is to do with this issue?

Mortific, 2021-09-17:
Is the gps reciever set to NMEA output? If you run gpsmon, what does it report?

msfu101, 2021-11-06:
No it's not,
the output of gpsmon is

tcp://localhost:2947          JSON slave driver>
(82) {"class":"VERSION","release":"3.20","rev":"3.20","proto_major":3,"proto_minor":14}
(32) {"class":"DEVICES","devices":[]}
(122) {"class":"WATCH","enable":true,"json":false,"nmea":false,"raw":2,"scaled":false,"timing":false,"spl
it24":false,"pps":true}

I don't know what i am doing wrong.

msfu101, 2021-11-07:
if i run sudo ./crocodilehunter.py -d -w
i get this output 

03:24:52 default - INFO starting crocodile hunter project: default
 * 03:24:52 default - INFO Starting WebUI
 * 03:24:52 default - INFO  * Running on http://0.0.0.0:5000/ (Press CTRL+C to quit)
 * 03:24:52 default - INFO Checking for config file
 * 03:24:52 default - INFO Compiling srsUE
 * 03:24:53 default - INFO SrsUE built succesfully
 * 03:24:53 default - INFO Reticulating Splines
 * 03:24:53 default - DEBUG Connecting to gpsd socket at 127.0.0.1:2947
 * 03:24:53 default - DEBUG Waiting for welcome message
 * 03:24:53 default - DEBUG Enabling gps
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - INFO No GPS fix detected. Waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - DEBUG Try %s waiting for fix.
 * 03:24:55 default - DEBUG Polling gps
 * 03:24:55 default - CRITICAL No GPS fix detected. Exiting.
 * 03:24:55 default - ERROR Unhandled exception:
Traceback (most recent call last):                                                                       
  File "./crocodilehunter.py", line 281, in <module>                                                     
    crocodile_hunter.start()                                                                             
  File "./crocodilehunter.py", line 137, in start                                                        
    exit(1)                                                                                              
  File "/usr/lib/python3.8/_sitebuiltins.py", line 26, in __call__                                       
    raise SystemExit(code)                                                                               
SystemExit: 1                                                                                            
                                                                                                         
 * 03:24:55 default - ERROR Exiting...
 * 03:24:55 default - WARNING Stopping Watchdog
 * 03:24:55 default - SUCCESS See you space cowboy...

so what can i do to get the gps fix?

msfu101, 2021-11-07:
when i run
/usr/lib/uhd/utils$ sudo ./query_gpsdo_sensors
it says this

Creating the USRP device with: ...
[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
Using Device: Single USRP:
  Device: B-Series Device
  Mboard 0: B210
  RX Channel: 0
    RX DSP: 0
    RX Dboard: A
    RX Subdev: FE-RX2
  RX Channel: 1
    RX DSP: 1
    RX Dboard: A
    RX Subdev: FE-RX1
  TX Channel: 0
    TX DSP: 0
    TX Dboard: A
    TX Subdev: FE-TX2
  TX Channel: 1
    TX DSP: 1
    TX Dboard: A
    TX Subdev: FE-TX1

gps_locked sensor not found.  This could mean that you have not installed the GPSDO correctly.

Visit one of these pages if the problem persists:
 * N2X0/E1X0: http://files.ettus.com/manual/page_gpsdo.html
 * X3X0: http://files.ettus.com/manual/page_gpsdo_x3x0.html
anyone an idea? i can*t get it

Mortific, 2021-11-07:
It seems like the gpds service is not getting a position from the gps.

1. Do you have a GPS receiver with usb? Is there a way to verify that the gps is getting a signal and gps fix. In my case when the led flashes on the GPS receiver, I have a gps fix. Fixed light means operating but no signal.

2. Have you installed and started the gpsd sevice according to installation instructions?

3. Is [device] in the config file pointed to the right USB port connected to the gps?

Regards
/Mort

codewith7us, 2022-11-01:
@msfu101 @Mortific is this problem solved ? i am getting same as @msfu101 when i run /usr/lib/uhd/utils$ sudo ./query_gpsdo_sensors

Regards

msfu101, 2024-01-08:
Problem solved. It was a wrong configured gpsd.