EFForg/crocodilehunter #131: Error creating objects in the database

cguerrave, 2022-01-24:
Related to #122 , the new fields (api_key, ext_id, uploaded) are not nullable, and this is making it impossible to store tower data in the database, I tested making the fields nullable, and the information is uploaded, not sure if this behavior is expected, the program is sending None/Null to the database for these 3 fields, even when api_client.py was executed successfully, and a key is available in config.ini

closed 2022-02-14 by cooperq

cooperq, 2022-03-07:
I think this is fixed now

alphafox02, 2022-06-16:
I’m not sure what was addressed here or how it was fixed, but finally after all this time I got back around to setting up a client / server connection. All code is the latest, database looks good, and with the server started up I got a good key generation by the client api. 

I then use the client api to add towers but in terminal I see that nothing is sent despite the local client having tower data in its database. I jump over to the server side and check the project name/tables etc and show no entries. How was this resolved besides doing what the author of the ticket suggests? I’ll try and see if what’s mentioned above works.

alphafox02, 2022-06-16:
I think I see what the person who submitted the ticket means now. Upon creating a new database with a -p and then trying to run Croc Hunter I’m unable to submit/enter info into the local MySQL server database. 

There’s a lengthy OperationalError, but it states (1048, “Column ‘api_key’ cannot be null” and so the insert into Tower data fails.

keldnorman, 2022-09-18:
I have the same problem on a clean installed Ubuntu 20.04 and fresh git clone of the repo from 18 sep 2022

cooperq, 2022-10-25:
Okay I finally looked into this. For some reason I made the api_key and some other fields "NOT NULL" which is not how it should be. I have fixed that and now it should all work. You might need to recreate your database or modify it manually for those changes to take effect. Fixed in https://github.com/EFForg/crocodilehunter/commit/44298d059dcc5c7c85ceb79ce21d8d2f106265da