EFForg/crocodilehunter #141: RTL-SDR compatibility?

Spessmen14, 2022-11-04:
I absolutely cannot drop 400 on a limesdr, does the RDL-SDR dongle work with this project?

Mortific, 2022-11-04:
I would dare to say no. According to technical specifications rtl-sdr goes up to 1.75 GHz. Over here it would be useless at the moment as LTE uses 2.3-2.5 GHz. And then we all the huzzle with libraries and dependencies etc etc.

imnikola, 2025-08-15:
A question, seeing that limesdr is menrioned. Is there support for it?

cooperq, 2025-08-15:
This project is no longer supported. Thanks

Cooper Quintin
Senior Staff Technologist
EFF.org is member supported. Join today!<https://www.eff.org/donate>

Signal: hunter.02
Find me on bluesky @cooperq.com or  @***@***.*** exchange on mastodon